TreeFallSound/pi-stomp #3: ili9341 TFT issues

micahvdm, 2020-10-18:
Hi, I’ve tried the default SPI pins for the ili9341 TFT on pi Stomp, but can’t seem to get anything to show. I did clone the repo and changed the display in modalapistomp.py but nothing changed. Any ideas?

rreichenbach, 2020-10-18:
First, know that the LCD stubs in the pistomp module other than lcdgfx.py are not fully implemented yet, but offered as starting points for support of other displays.  As I remember, this driver does display the current pedalboard name along with 3 hardcoded colored squares which I had planned to used for representing effects in the pedalboard.

I would suggest checking your pinout.
The for the ili9341 driver, I was using this specific display (https://www.amazon.com/gp/product/B01CZL6QIQ/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1).  You can alter the specific GPIO pins in lcdili9341.py.  But here is what I used:
CS=BCM8=pin24
Reset=BCM5=pin29
DC=BCM6=pin31
MOSI=BCM10=pin19
SCLK=BCM11=pin23

LED=5V=pin2 or 4
VCC=3V3=pin1 or 17
GND=pin6 or 9 or 14 or 20 or 23 or 30 or 34 or 39

micahvdm, 2020-10-19:
I have the same display, used the same pins, but still nothing. I will start working on porting the code from lcdgfx.py to lcdili9341.py soon. If you'd like, when I have it working like the gfxhat, I can do a PR. Just let me know.

rreichenbach, 2020-10-19:
Is the display backlight illuminating?  If not, you won't see anything.  Simply connecting the LED, VCC and ground should enable that much and you'll see an all white screen.

If that's working, and you're confident about your other GPIO connections, it's possible that you're hitting a software issue.  That could be because you don't have other hardware connected.  The LCD is initialized after the soundcard and midi.  So you might want to comment out from line 54 thru 72 in main() in modalapistomp.py along with lines 77 thru 95 and 101.   Alternatively, move line 75 to the top of main().

Also invoke modalapistomp.py from the command line with "-l debug" so you can see any debug messages going to stdout/stderr.

Here's what it should like when you get it working:
https://treefallsound.com/wiki/lib/exe/fetch.php?media=tft-lcd-test.jpg

micahvdm, 2020-10-21:
I have gotten the display working now. It was not initializing due to my audio interface failing to load. Need to now rewrite the lcdgfx.py code across to lcdili9341.py. Python is relatively new for me, but I don't mind the challenge.

rreichenbach, 2020-10-26:
Cool.  FWIW, the lcdgfx.py is the least portable in the codebase.  It has lots of hardcoded numbers, etc. because it needs to render the pixels in specific "zones" of the 128x64 LCD.  But we at least have the abstract interface defined by the lcd.py base class.  Flesh those methods out one at a time and you should have something usable.  My vision was that each display type could render the same info completely differently.  A color LCD, has the benefit of an extra "dimension".  So for example it could use color or fill to indicate effect on/off instead of the "bar" decorator used in lcdgfx.py.  Was also thinking that color could be used to imply plugin category (Green for Distortion, Blue for Modulation, etc.)

micahvdm, 2020-11-26:
I have decided to rather order the same display that you are using so that I can focus on adding features to the device. Sound card arrives today and display will arrive hopefully next week. Have you tried accessing the mod tuner to display btw? Would be super great if we can implement that!!

rreichenbach, 2020-11-26:
Cool.  Yeah, I've thought about the tuner.  It would need to have real-time code specific to the tuner plugin.  I created an API for getting LV2 plugin settings (http://lv2plug.in/ns/lv2core#InputPort), but it doesn't work for dynamic values (http://lv2plug.in/ns/lv2core#OutputPort).  Not sure what might be required.  Might get some clues from the Tuna Tuner plugin.  Also not sure how responsive it will be, Rest calls (although local) might be too slow.  I don't have high hopes as I feel Tuna Tuner in the MOD webui is pretty sluggish and not something I would actually use. 

Would you mind if we moved discussions like this to the pi-Stomp forum?  https://treefallsound.com/forum/  Not that there's much traffic there yet, but I feel discussions which aren't completely software related belong there, and it might eventually get more visibility.

rreichenbach, 2021-01-09:
Many stubs exist for various displays.  Improvements were made last month to make it easier to support new displays.  Closing this issue, if specific issues are found please file a new one.

closed 2021-01-09 by rreichenbach