TreeFallSound/pi-stomp #7: First Boot, no display

KevWal, 2020-12-24:
Hi

I followed the instructions here:

"https://www.treefallsound.com/wiki/doku.php?id=start#building_pi_stomp"

to the letter as far as I am aware, but on third boot (first boot just configures my home wifi, second boot runs ./setup.sh which looked to go well) I got no display.

I had the following errors in dmesg:
```
[    8.063443] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    8.063471] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    8.202134] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    8.202164] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    8.517447] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[    8.660398] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[    8.832363] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    8.832392] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    8.834051] brcmfmac: F1 signature read @0x18000000=0x15264345
[    8.848816] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    8.851792] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[    8.851815] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[    8.880274] usbcore: registered new interface driver brcmfmac
[    9.130063] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.150351] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Mar  2 2020 23:30:41 version 7.45.202 (r724630 CY) FWID 01-72f6ece2
[   10.137247] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[   10.137275] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   10.155557] audioinjector-stereo soc:sound: ASoC: CODEC DAI wm8731-hifi not registered - will retry
[   10.155584] audioinjector-stereo soc:sound: snd_soc_register_card failed (-517)
[   10.425902] wm8731 1-001a: Assuming static MCLK
[   10.425953] wm8731 1-001a: 1-001a supply AVDD not found, using dummy regulator
[   10.426343] wm8731 1-001a: Linked as a consumer to regulator.0
[   10.426378] wm8731 1-001a: 1-001a supply HPVDD not found, using dummy regulator
[   10.430926] wm8731 1-001a: 1-001a supply DCVDD not found, using dummy regulator
[   10.431214] wm8731 1-001a: 1-001a supply DBVDD not found, using dummy regulator
[   10.439437] audioinjector-stereo soc:sound: wm8731-hifi <-> 3f203000.i2s mapping ok
[   16.472023] random: crng init done
```

I finally ran this command:

"/usr/bin/python3 /home/patch/pi-stomp/modalapistomp.py --log debug"

that then completed the hardware test successfully and exited.  I then rebooted, but no display shown again.

I now have the file hardware test file:

patch@patchbox:~ $ ls -la ~/pi-stomp/pistomp/.hardware_tests_passed
-rw-r--r-- 1 root root 0 Dec 24 13:37 /home/patch/pi-stomp/pistomp/.hardware_tests_passed

In /var/log/daemon.log pisound-ctl seems to be continually restarting:

`
Dec 24 13:47:54 patchbox pisound-ctl[899]: Reading Pisound version failed, did the kernel module load successfully?
Dec 24 13:47:54 patchbox pisound-ctl[899]: [899] main: Pisound Control Server 1.03 starting...
Dec 24 13:47:54 patchbox systemd[1]: pisound-ctl.service: Main process exited, code=exited, status=234/n/a
Dec 24 13:47:54 patchbox systemd[1]: pisound-ctl.service: Failed with result 'exit-code'.
Dec 24 13:47:59 patchbox systemd[1]: pisound-ctl.service: Service RestartSec=5s expired, scheduling restart.
Dec 24 13:47:59 patchbox systemd[1]: pisound-ctl.service: Scheduled restart job, restart counter is at 56.
Dec 24 13:47:59 patchbox systemd[1]: Stopped Bluetooth service for Pisound companion app..
Dec 24 13:47:59 patchbox systemd[1]: Started Bluetooth service for Pisound companion app..
Dec 24 13:47:59 patchbox pisound-ctl[901]: Reading Pisound version failed, did the kernel module load successfully?
Dec 24 13:47:59 patchbox pisound-ctl[901]: [901] main: Pisound Control Server 1.03 starting...
Dec 24 13:47:59 patchbox systemd[1]: pisound-ctl.service: Main process exited, code=exited, status=234/n/a
Dec 24 13:47:59 patchbox systemd[1]: pisound-ctl.service: Failed with result 'exit-code'.
Dec 24 13:48:04 patchbox systemd[1]: pisound-ctl.service: Service RestartSec=5s expired, scheduling restart.
Dec 24 13:48:04 patchbox systemd[1]: pisound-ctl.service: Scheduled restart job, restart counter is at 57.
Dec 24 13:48:04 patchbox systemd[1]: Stopped Bluetooth service for Pisound companion app..
Dec 24 13:48:04 patchbox systemd[1]: Started Bluetooth service for Pisound companion app..
Dec 24 13:48:04 patchbox pisound-ctl[948]: Reading Pisound version failed, did the kernel module load successfully?
Dec 24 13:48:04 patchbox pisound-ctl[948]: [948] main: Pisound Control Server 1.03 starting...
Dec 24 13:48:04 patchbox systemd[1]: pisound-ctl.service: Main process exited, code=exited, status=234/n/a
Dec 24 13:48:04 patchbox systemd[1]: pisound-ctl.service: Failed with result 'exit-code'.
`

Running modalapistomp.py manuall then turned the LCD on, but the exited:
`
root@patchbox:/var/log# /usr/bin/python3 /home/patch/pi-stomp/modalapistomp.py --log debug
Log level now set to: DEBUG
DEBUG:rtmidi.midiutil:Creating MidiOut object.
INFO:rtmidi.midiutil:Opening MIDI output port #0 (Midi Through:Midi Through Port-0 14:0).
INFO:root:Init mod
DEBUG:root:Init hardware
DEBUG:root:Relay on: 12
DEBUG:urllib3.connectionpool:Starting new HTTP connection (1): localhost:80
ERROR:root:Cannot connect to mod-host
`
The root of my issues seems to be jack.service not running:
'
root@patchbox:/var/log# systemctl status jack.service 
● jack.service - JACK Server
   Loaded: loaded (/lib/systemd/system/jack.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-12-24 13:52:53 GMT; 38s ago
  Process: 1283 ExecStart=/etc/jackdrc (code=exited, status=255/EXCEPTION)
 Main PID: 1283 (code=exited, status=255/EXCEPTION)

Dec 24 13:52:53 patchbox jackdrc[1283]: creating alsa driver ... hw:ALSA|hw:ALSA|128|2|48000|0|0|nomon|swmeter|soft-mode|16bit
Dec 24 13:52:53 patchbox jackdrc[1283]: control open "hw:ALSA" (No such device)
Dec 24 13:52:53 patchbox jackdrc[1283]: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
Dec 24 13:52:53 patchbox jackdrc[1283]: ALSA lib pcm_hw.c:1822:(_snd_pcm_hw_open) Invalid value for card
Dec 24 13:52:53 patchbox jackdrc[1283]: ALSA: Cannot open PCM device alsa_pcm for playback. Falling back to capture-only mode
Dec 24 13:52:53 patchbox jackdrc[1283]: Cannot initialize driver
Dec 24 13:52:53 patchbox jackdrc[1283]: JackServer::Open failed with -1
Dec 24 13:52:53 patchbox jackdrc[1283]: Failed to open server
Dec 24 13:52:53 patchbox systemd[1]: jack.service: Main process exited, code=exited, status=255/EXCEPTION
Dec 24 13:52:53 patchbox systemd[1]: jack.service: Failed with result 'exit-code'.
'
I had to "reconfigure Jack using sudo patchbox-config and going to ‘jack’ section" which then seemed to get it running.

Another reboot and things look a lot better, http service running, screen showing settings for Distortion pedal, etc.
Hopefully I am all working ok now, but something isn't working corretly in the setup process and debuging was hard - and I am familiar with Linux...

Hope the writeup helps find the issues.

Cheers
Kev

rreichenbach, 2020-12-25:
Thanks for your report.

I just tried a fresh install to make sure it's not a problem with external code sources changing recently.  It booted just fine after hardware test.  Not saying there's still not a bug in the setup scripts though.

I don't believe pisound-ctl failing or restarting should cause a problem.  In fact stopping it should be fine (sudo systemctl stop pisound-ctl;  sudo systemctl disable pisound-ctl).  That should probably be added to the setup scripts.

Jack not starting though is most likely an issue either with the jack config or with the audioinjector hardware or driver.  Once you got Jack started, did it pass audio?  If not it's probably best to dive into why.  I'd first run 'arecord -l' and 'aplay -l'.  Both of those should identify the audioinjector-pi-soundcard as 'device 0'.  If that's not the case, try running 'alsa-info --stdout'.  That will include the dmesg log which may complain about the codec not registered for audioinjector, that's expected as long as it finally finds it and says "mapping ok".

KevWal, 2020-12-30:
Thanks, no guitar here at the moment, will test audio shortly.  Seems all ok now through apart from not being able to test audio.

rreichenbach, 2021-01-09:
Any update @KevWal ?

KevWal, 2021-01-10:
Hi Randy

My son is now back at Uni with the Stomp and his guitar, initial tests seem all good, he says he is a bit confused by some of the rotary encoder and push button actions - but that is probably down to him being unfamiliar with the complexity of the stomp.

Thanks
Kevin

rreichenbach, 2021-04-04:
Issue appears to be resolved

closed 2021-04-04 by rreichenbach