TreeFallSound/pi-stomp #8: Very low and noisy volume on OUT1

gallochri, 2021-04-03:
As from the title.
I did various tests, I also tried to replace the sound card but on OUT1 the volume remains very low and background noises can be heard.
By lowering the volume even slightly from the maximum value, the sound disappears, both using IN1 or IN2.
How can I check if there is any faulty component? is it possible to have the wiring diagram?
No problem on OUT2 instead.

rreichenbach, 2021-04-03:
Please check out the troubleshooting guide:
https://treefallsound.com/wiki/doku.php?id=troubleshooting_guide

Not quite understanding your statement "the sound disappears".  Is that the signal you want to hear that disappears, or the noise?  When you say "No problem on OUT2", does that mean it has full output and no noise?

Also helpful to know:
1) What kind of noise you're hearing.  50/60Hz hum?  Hiss?  Digital noise?

2) Do you get full signal or noise when processing is bypassed (footswitch #1 white halo is off)?

3) If you unplug both input jacks and load a pedalboard like "SEQUENCED_DRUMS", do you hear drum beats in both outputs? Is it noisy?  If so, that could more point toward a output side problem instead of input side.

I'm attaching the Audio portion of the pi-Stomp 1.0.2 schematic.  Yellow highlights the signal path from the IN1 jack (J6-IL) to the audio card via the relay and buffer.  Blue highlights the path from the audio card to the output jack via the Volume pot and relay.  Note that when bypassed, the relay simply connects the input jack straight thru to the output jack (via the N.C. contacts).

In2/Out2 works just the same.

[image]

gallochri, 2021-04-04:
> Is that the signal you want to hear that disappears, or the noise? 

Only the signal, the noise remains constant.

> When you say "No problem on OUT2", does that mean it has full output and no noise?

Yes, full output and no noise.

> What kind of noise you're hearing. 50/60Hz hum? Hiss? Digital noise?

Hiss and digital noise.

> Do you get full signal or noise when processing is bypassed (footswitch #1 white halo is off)?

With bypass on (footswitch # 1 off) full output on Out1 and no output on Out2.

> If you unplug both input jacks and load a pedalboard like "SEQUENCED_DRUMS", do you hear drum beats in both outputs? Is it noisy? If so, that could more point toward a output side problem instead of input side.

Yes, the problem remains the same, Out2 works perfectly, while the problems described remain on Out1.

I try to reopen the pedal and check the connections. 
From what I understand from the wiring diagram I think the problem is with the potentiometer or the relay, right?

gallochri, 2021-04-04:
I replaced the Volume pot (probably damaged during assembly, my fault) and now everything works fine.
Many thanks for the support.

closed 2021-04-04 by gallochri