TreeFallSound/pi-stomp #9: Change Pedalboard via MIDI

gallochri, 2021-04-12:
Is it possible to change pedalboard via midi?

rreichenbach, 2021-04-12:
Yes.  Program Change messages coming into the MIDI port should change the pedalboard active in MOD.  The bad news is that the pi-Stomp software (separate from MOD) doesn't currently listen to incoming MIDI messages, so the LCD will not reflect the pedalboard change.  It certainly could.  Feel free to submit an issue to request adding that functionality.

gallochri, 2021-04-17:
I can't find which channel to send "Program Change" messages to change pedalboards.
Is there a log of incoming midi messages?

rreichenbach, 2021-04-18:
pi-stomp specific software doesn't log messages coming in, only the messages it generates.  You could likely use a midi utility like amidi with the dump option, but that doesn't seem to work for me.  Also, I'm not seeing Mod respond to PC messages.  It did at one time, but it seemed they might have changed that or there's some setting required to enable it.

ozbenh, 2021-11-01:
Talking of which, when the board is changed via the web UI, this is also not reflected in pi-stomp... should we also expect a MIDI message ? Otherwise is there a way to poll MOD for the current board and adapt ?

rreichenbach, 2021-11-02:
There is no MIDI message sent when the pedalboard is changed via MOD UI.   And the communication between pi-stomp software and mod-ui/mod-host has been unidirectional via the rest API defined by mod-ui/webserver.py (woefully incomplete BTW).  For their own hardware, MOD does register callbacks for keeping the hardware in sync with the MOD UI.  I've had plans to do something similar, but haven't invested the time.  But yes, it's not very nice that at least pedalboard and preset changes from MOD UI are not synced on the pi-stomp.   A few months ago, I did create a polling based solution for pedalboard changes.  I've not discovered a similar hook for preset/snapshot changes, but I suppose pedalboard changes would be better than nothing.  I'll find and post that change once I find the SD card that has it (never pushed it).