TreeFallSound/pi-stomp #27: pisound-btn coming back

ozbenh, 2021-11-08:
This is for keeping track of the discussion we had on the forums. Updates to pisound-btn will cause it to re-enable
itself which we don't want.

Blokas has now merged my pull requests, so any time soon, we should see pisound-btn 1.15 and patchbox-wifi 1.0.0 hitting the repos (well, I hope...)

When they show up, we should update the setup/update script to ensure we pull in patchbox-wifi and remove the pisound-btn package completely (and pisound-ctl while at it).

ozbenh, 2021-11-08:
The updates packages  are in the patchbox beta channel, see: https://github.com/BlokasLabs/patchbox-os-debs/pull/2

Testing welcome by others than just me

rreichenbach, 2025-06-05:
Closing this as pi-Stomp hasn't been patchbox based for years.

closed 2025-06-05 by rreichenbach