TreeFallSound/pi-stomp #34: No creation of plugin box on the pistomp lcd when a footswitch is connected to any function.

dormirj, 2022-03-22:
Currently, when a footswitch is connected to any function of a plugin in the pedalboard configuration it will not be visible as a box with a name in the pistomp ui on the screen. This is fine if the footswitch controls the bypass button of the plugin, but creates an issue when the footswitch is connected to any other function on the pedal (e.g. tap-tempo, loop record, etc.), because then you can't bypass the plugin from the unit.

A fix for me would be the option to show all plugins not depending on whether they are connected to a footswitch or not, or, probably more involved, to check if the footswitch controls the bypass or something else.

rreichenbach, 2024-10-04:
Fixed via commit: 59633a99b8d74dc80e33ade7ba84328208b88d04

Available on the beta-bookworm-dev branch tag: v2.3.0-beta.1

rreichenbach, 2024-10-04:
Closed per previous comment/fix

rreichenbach, 2024-10-04:
Note that the fix above is for v2 and v3 hardware, not v1.

closed 2024-10-04 by rreichenbach