TreeFallSound/pi-stomp #52: pi stomps keeps restarting

fcebreros, 2023-01-16:
Hi!
I updated the the software to the latest version and was editing the configuration file.
I tried the switch without closing the files and it seems I somehow introduced a bug
because it restarted and keeps restarting no end after that. 
Here is the log:
```
 mod-ala-pi-stomp.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 17:28:12 patchbox systemd[1]: mod-ala-pi-stomp.service: Failed with result 'exit-code'.
Jan 16 17:28:14 patchbox systemd[1]: mod-ala-pi-stomp.service: Service RestartSec=2s expired, scheduling restart.
Jan 16 17:28:14 patchbox systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 102.
Jan 16 17:28:14 patchbox systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 16 17:28:14 patchbox systemd[1]: Started MOD-ALA-PI-STOMP.
Jan 16 17:28:15 patchbox python3[5319]: Traceback (most recent call last):
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/modalapistomp.py", line 149, in <module>
Jan 16 17:28:15 patchbox python3[5319]:     main()
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/modalapistomp.py", line 86, in main
Jan 16 17:28:15 patchbox python3[5319]:     hw = factory.create(handler, midiout)
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/hardwarefactory.py", line 37, in create
Jan 16 17:28:15 patchbox python3[5319]:     return Pistompcore.Pistompcore(self.cfg, handler, midiout, refresh_callback=handler.update_lcd_fs)
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/pistompcore.py", line 81, in __init__
Jan 16 17:28:15 patchbox python3[5319]:     self.reinit(None)
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/hardware.py", line 81, in reinit
Jan 16 17:28:15 patchbox python3[5319]:     self.__init_footswitches(self.cfg)
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/hardware.py", line 227, in __init_footswitches
Jan 16 17:28:15 patchbox python3[5319]:     fs.add_relay(self.relay)
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/footswitch.py", line 118, in add_relay
Jan 16 17:28:15 patchbox python3[5319]:     self.set_value(not relay.init_state())
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/relay.py", line 43, in init_state
Jan 16 17:28:15 patchbox python3[5319]:     self.disable()
Jan 16 17:28:15 patchbox python3[5319]:   File "/home/patch/pi-stomp/pistomp/relay.py", line 67, in disable
Jan 16 17:28:15 patchbox python3[5319]:     shutil.chown(f, user="pistomp", group=None)
Jan 16 17:28:15 patchbox python3[5319]:   File "/usr/lib/python3.7/shutil.py", line 1059, in chown
Jan 16 17:28:15 patchbox python3[5319]:     raise LookupError("no such user: {!r}".format(user))
Jan 16 17:28:15 patchbox python3[5319]: LookupError: no such user: 'pistomp'
Jan 16 17:28:16 patchbox systemd[1]: mod-ala-pi-stomp.service: Main process exited, code=exited, status=1/FAILURE
Jan 16 17:28:16 patchbox systemd[1]: mod-ala-pi-stomp.service: Failed with result 'exit-code'.
Jan 16 17:28:18 patchbox systemd[1]: mod-ala-pi-stomp.service: Service RestartSec=2s expired, scheduling restart.
Jan 16 17:28:18 patchbox systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 103.
Jan 16 17:28:18 patchbox systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 16 17:28:18 patchbox systemd[1]: Started MOD-ALA-PI-STOMP.
```

rreichenbach, 2023-01-17:
I assume to get that log you were logged in as user pistomp, correct?  Seems rather impossible that there would be "no such user" then.

You could run this:
cat /etc/passwd | grep pistomp

If that returns a line of info then pistomp does exist.   In that case, I would just move the ~/pi-stomp directory aside, clone a fresh one and see if that fixes the problem:

```
ps-stop
mv ~/pi-stomp ~/pi-stomp0
git clone
https://github.com/TreeFallSound/pi-stomp.git
ps-restart
```
If everything seems to work, you could then remove ~/pi-stomp0 since ~/pi-stomp is your working pi-stomp dir now.

However if the passwd file doesn't contain a pistomp user, then yeah, some corruption occurred.  You could possibly fix it but I feel it would be more reliable to do a fresh install.

fcebreros, 2023-01-17:
Hi,
I was logged in as patch.

rreichenbach, 2023-01-17:
Oh.  So apparently you did a 'git pull'?  That'll update the pi-stomp specific software, but there is no patch user on the new mainline (2.1 and beyond).  It's no longer Patchbox based.  The last Patchbox based build is branch: v2.0.2-patchboxOS-based.  If you want that run:
git clone -b v2.0.2-patchboxOS-based https://github.com/TreeFallSound/pi-stomp.git

Don't know what version of pi-Stomp/patchbox you were on, but the improvements wouldn't be near as significant as updating to the new non-Patchbox mainline.  It includes a way more recent version of MOD, is 64-bit, includes the realtime kernel and tons more great plugins (650+).

That requires a build from scratch: reimaging your SD card.  Your old pedalboards can be saved off first and reinstalled.  Instructions for all of that are here:
https://www.treefallsound.com/wiki/doku.php?id=software_installation_64-bit

fcebreros, 2023-01-18:
OK, thanks. 
I did a new build from scratch.

rreichenbach, 2023-01-18:
Closing issue.  Adding Trello item for tracking a code change to avoid such issues.

closed 2023-01-18 by rreichenbach