TreeFallSound/pi-stomp #54: UI restarts when trying to connect with browser

fcebreros, 2023-01-20:
Hi,
I dont get any audio  processed (bypass sounds OK)
and when I try to connect to the pistomp with a browser
it starts to load the patch on the screen and then
looses connection and screen on the pistomp reloads.
I get this error and I'm not sure
if it's related to the audio problem.
Thanks
Frank
```
-- Journal begins at Wed 2022-09-21 22:06:10 -05. --
Jan 20 18:19:39 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 20 18:19:39 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 11.116s CPU time.
Jan 20 18:19:39 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
Jan 20 18:19:43 pistomp python3[5472]: ERROR:root:Cannot connect to mod-host
Jan 20 18:19:44 pistomp systemd[1]: mod-ala-pi-stomp.service: Succeeded.
Jan 20 18:19:44 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.441s CPU time.
Jan 20 18:19:46 pistomp systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 1.
Jan 20 18:19:46 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 20 18:19:46 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.441s CPU time.
Jan 20 18:19:46 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
Jan 20 18:20:32 pistomp systemd[1]: Stopping MOD-ALA-PI-STOMP...
Jan 20 18:20:33 pistomp systemd[1]: mod-ala-pi-stomp.service: Succeeded.
Jan 20 18:20:33 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 20 18:20:33 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 7.021s CPU time.
Jan 20 18:20:33 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
Jan 20 18:20:38 pistomp python3[5673]: ERROR:root:Cannot connect to mod-host
Jan 20 18:20:39 pistomp systemd[1]: mod-ala-pi-stomp.service: Succeeded.
Jan 20 18:20:39 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.424s CPU time.
Jan 20 18:20:41 pistomp systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 1.
Jan 20 18:20:41 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
Jan 20 18:20:41 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.424s CPU time.
Jan 20 18:20:41 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
```

rreichenbach, 2023-01-20:
I'm a bit confused.  Last I knew you were modifying the configuration file.  That's usually something one would do only after basic functionality (like audio) is working. 

Also, that log file is from 4 months ago.

Please review the Troubleshooting guide:
https://www.treefallsound.com/wiki/doku.php?id=troubleshooting_guide

If that doesn't help then let me know at:
support@treefallsound.com

I'll need to understand your hardware situation, and any software modifications you've made.  Email is much better for that.

rreichenbach, 2023-06-20:
Closing until the original problem can be reproduced

closed 2023-06-20 by rreichenbach