TreeFallSound/pi-stomp #57: MIDI learn only working for existing pedalboards

MarnixMeersman, 2023-03-04:
Hi,

I have a problem where I can only assign potentiometers and switches via MIDI-learn when I'm in the pedalboards that came with the installation. If I make a new pedalboards and save it on my pistomp, (I verified it was saved since I could play through it without the pi connected to WiFi) I'm unable to assign any midi control. 

I went through every option in the troubleshooting guide on the tree fall sound website. Nothing worked. 

Hoping someone can help. 

Cheers Marnix

rreichenbach, 2023-03-05:
It sounds like your new pedalboard has the MIDI mode set to Aggregated.  For MIDI learn to work, it should be set to Segregated.

See this tutorial on how to change modes:
https://treefallsound.com/wiki/doku.php?id=tutorials_and_videos#controlling_pi-stomp

It's possible that the latest version of MOD has changed the default mode to be Aggregated.  I'll need to investigate.  Until then, try the manual mode change.

MarnixMeersman, 2023-03-06:
Thanks, that was indeed the problem. Very happy with my pistomp. As promised over email, a picture of my build (will still get a paintjob). 
[image]

rreichenbach, 2023-03-07:
Wow, awesome!  That's one maxxed out pi-Stomp!  Please share on the forum!  (https://treefallsound.com/forum)

micahvdm, 2023-05-21:
MIDI mapping is fixed with the latest PR. Please do a full reinstall to fix!

closed 2023-06-20 by rreichenbach