TreeFallSound/pi-stomp #61: LCD does not start with fresh installation on hardware version 1

gallochri, 2023-05-10:
`May 10 20:56:40 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
May 10 20:56:40 pistomp sudo[9459]: pam_unix(sudo:session): session closed for user root
May 10 20:56:42 pistomp jackdrc[440]: port created: RtMidiOut-Client:midi/playback_1
May 10 20:56:42 pistomp python3[9462]: Traceback (most recent call last):
May 10 20:56:42 pistomp python3[9462]:   File "/home/pistomp/pi-stomp/modalapistomp.py", line 153, in <module>
May 10 20:56:42 pistomp python3[9462]:     main()
May 10 20:56:42 pistomp python3[9462]:   File "/home/pistomp/pi-stomp/modalapistomp.py", line 90, in main
May 10 20:56:42 pistomp python3[9462]:     hw = factory.create(handler, midiout)
May 10 20:56:42 pistomp python3[9462]:   File "/home/pistomp/pi-stomp/pistomp/hardwarefactory.py", line 35, in create
May 10 20:56:42 pistomp python3[9462]:     return Pistomp.Pistomp(self.cfg, handler, midiout, refresh_callback=handler.update_lcd_fs)
May 10 20:56:42 pistomp python3[9462]:   File "/home/pistomp/pi-stomp/pistomp/pistomp.py", line 84, in __init__
May 10 20:56:42 pistomp python3[9462]:     self.init_lcd()
May 10 20:56:42 pistomp python3[9462]:   File "/home/pistomp/pi-stomp/pistomp/pistomp.py", line 97, in init_lcd
May 10 20:56:42 pistomp python3[9462]:     self.mod.add_lcd(Lcd.Lcd(self.mod.homedir))
May 10 20:56:42 pistomp python3[9462]: TypeError: Can't instantiate abstract class Lcd with abstract method update_eq
May 10 20:56:42 pistomp systemd[1]: mod-ala-pi-stomp.service: Main process exited, code=exited, status=1/FAILURE`

micahvdm, 2023-05-11:
Did you install with -v 1.0 after setup.sh?

gallochri, 2023-05-11:
After a reboot this is the `journalctl -f`
```
May 11 12:52:35 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
May 11 12:52:37 pistomp python3[777]: Traceback (most recent call last):
May 11 12:52:37 pistomp python3[777]:   File "/home/pistomp/pi-stomp/modalapistomp.py", line 29, in <module>
May 11 12:52:37 pistomp python3[777]:     import pistomp.testhost as Testhost
May 11 12:52:37 pistomp python3[777]:   File "/home/pistomp/pi-stomp/pistomp/testhost.py", line 16, in <module>
May 11 12:52:37 pistomp python3[777]:     from pistomp.hardware import Hardware
May 11 12:52:37 pistomp python3[777]:   File "/home/pistomp/pi-stomp/pistomp/hardware.py", line 25, in <module>
May 11 12:52:37 pistomp python3[777]:     import pistomp.ledstrip as Ledstrip
May 11 12:52:37 pistomp python3[777]:   File "/home/pistomp/pi-stomp/pistomp/ledstrip.py", line 18, in <module>
May 11 12:52:37 pistomp python3[777]:     import matplotlib
May 11 12:52:37 pistomp python3[777]: ModuleNotFoundError: No module named 'matplotlib'
May 11 12:52:37 pistomp systemd[1]: mod-ala-pi-stomp.service: Main process exited, code=exited, status=1/FAILURE
May 11 12:52:37 pistomp systemd[1]: mod-ala-pi-stomp.service: Failed with result 'exit-code'.
May 11 12:52:37 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 1.905s CPU time.
May 11 12:52:39 pistomp systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 18.
May 11 12:52:39 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
May 11 12:52:39 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 1.905s CPU time.
```
Shall I try the setup again with the indicated branch?

micahvdm, 2023-05-11:
yes, I think it would be best to start over on v2.1.1-staging.
make sure that your setup command looks like this:
nohup ./setup.sh -a audioinjector-wm8731-audio -v 1.0 > setup.log | tail -f setup.log

The branches default to iqaudio codec soundcard, so you have to tell the setup to select audioinjector

gallochri, 2023-05-11:
All working with installation from scratch and branch v2.1.1-staging.

rreichenbach, 2023-06-28:
Fix also added to the master branch, so closing this issue.

closed 2023-06-28 by rreichenbach

gallochri, 2023-08-09:
```
Aug 09 20:51:19 pistomp systemd[1]: Started MOD-ALA-PI-STOMP.
Aug 09 20:51:21 pistomp kernel: hwmon hwmon1: Undervoltage detected!
Aug 09 20:51:21 pistomp jackdrc[411]: port created: RtMidiOut-Client:midi/playback_1
Aug 09 20:51:21 pistomp sudo[43747]:     root : PWD=/ ; USER=root ; COMMAND=/usr/bin/amixer sset AUX Jack unmute
Aug 09 20:51:21 pistomp sudo[43747]: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=0)
Aug 09 20:51:21 pistomp python3[43748]: amixer: Unable to find simple control 'AUX Jack',0
Aug 09 20:51:21 pistomp sudo[43747]: pam_unix(sudo:session): session closed for user root
Aug 09 20:51:22 pistomp python3[43739]: Traceback (most recent call last):
Aug 09 20:51:22 pistomp python3[43739]:   File "/home/pistomp/pi-stomp/modalapistomp.py", line 153, in <module>
Aug 09 20:51:22 pistomp python3[43739]:     main()
Aug 09 20:51:22 pistomp python3[43739]:   File "/home/pistomp/pi-stomp/modalapistomp.py", line 90, in main
Aug 09 20:51:22 pistomp python3[43739]:     hw = factory.create(handler, midiout)
Aug 09 20:51:22 pistomp python3[43739]:   File "/home/pistomp/pi-stomp/pistomp/hardwarefactory.py", line 35, in create
Aug 09 20:51:22 pistomp python3[43739]:     return Pistomp.Pistomp(self.cfg, handler, midiout, refresh_callback=handler.update_lcd_fs)
Aug 09 20:51:22 pistomp python3[43739]:   File "/home/pistomp/pi-stomp/pistomp/pistomp.py", line 84, in __init__
Aug 09 20:51:22 pistomp python3[43739]:     self.init_lcd()
Aug 09 20:51:22 pistomp python3[43739]:   File "/home/pistomp/pi-stomp/pistomp/pistomp.py", line 97, in init_lcd
Aug 09 20:51:22 pistomp python3[43739]:     self.mod.add_lcd(Lcd.Lcd(self.mod.homedir))
Aug 09 20:51:22 pistomp python3[43739]: TypeError: Can't instantiate abstract class Lcd with abstract method update_eq
Aug 09 20:51:22 pistomp systemd[1]: mod-ala-pi-stomp.service: Main process exited, code=exited, status=1/FAILURE
Aug 09 20:51:22 pistomp systemd[1]: mod-ala-pi-stomp.service: Failed with result 'exit-code'.
Aug 09 20:51:22 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.891s CPU time.
Aug 09 20:51:22 pistomp jackdrc[411]: port deleted: RtMidiOut-Client:midi/playback_1
Aug 09 20:51:24 pistomp systemd[1]: mod-ala-pi-stomp.service: Scheduled restart job, restart counter is at 1104.
Aug 09 20:51:24 pistomp systemd[1]: Stopped MOD-ALA-PI-STOMP.
Aug 09 20:51:24 pistomp systemd[1]: mod-ala-pi-stomp.service: Consumed 2.891s CPU time.
```
Regression on master branch

rreichenbach, 2023-08-14:
This should be fixed now on the master branch.  Tested on v1 hardware.