TreeFallSound/pi-stomp #65: TouchOSC and recording

MarnixMeersman, 2023-06-15:
Hi, 

I have the newest version of the pi-stom firmware installed. 
I would like to start using 2 features, only I haven't managed to get it to work. 

The first one is TouchOSC. I struggle to find documentation on this in the wiki (only the touchosc2midi.service, but no clue how to use it). Same for the hotspot tbh. Is the hotspot meant to control it with a computer without having a home network? If so, it would be nice to know how to set thuis up as well. 

Secondly, a big feature for me would be to be able to record myself within the MOD-UI. There seem to be a few pedals that seem to be specifically designed for this (see image). These pedals seem to function well, but I have no clue where the recording is stored or how to access it. 
[image]

It would be awesome if you could point me in the right direction o n this. 

Thanks!

rreichenbach, 2023-06-28:
Your question about TouchOSC inspired me to create a tutorial on how to set it up:
https://www.treefallsound.com/wiki/doku.php?id=tutorials_and_videos#using_touchosc

Hotspot mode is documented (maybe not greatly) here:
https://www.treefallsound.com/wiki/doku.php?id=pi-stomp_core#mod_web_ui_for_pedalboard_editing

Yes, Hotspot mode allows you to connect to the pi-Stomp from a computer/phone, etc without need for a home network.  In that mode, the pi-Stomp presents an add-hoc network that your device can connect to.

Regarding the recording plugins, if you hover over the "Rec" button, you should see the path where the file gets saved.  That should be the same folder you'll find on the "File Manager" page of the MOD UI.  Get to that page via the toolbar button in the bottom left of the UI.   However I'm not seeing that any file gets created.  That should I don't know if that's a problem with the plugin(s), MOD or our fork of MOD.  I will look into it.

MarnixMeersman, 2023-06-30:
Hi,

Thanks a lot. I can see how much effort you guys put into pi-stomp. Maybe I can contribute some day with some cool pedalboards for the MOD guitar synth. 

Regarding recording, I face the exact same problem. Even going to the exact path via SSH gives nothing so it might be an issue from the mod itself. Also, not sure if the processor can even handle this (live wave format recording and writing to the SSD while processing the signal chain sounds like a lot to handle for a pi) 

Thanks a lot for the tutorial and documentation again!