TreeFallSound/pi-stomp #89: USB midi output dies after a couple of notes

protectorateindustrial, 2023-10-16:
USB midi output dies after a while. Clock messages aslo affected. Midi message monitor lv2 displays the messages, nothing from the physical output. Tested with roland um-1 and iConnectivity mio.

rreichenbach, 2023-10-24:
We might need some more info.  So your um-1 or mio is connected to the pi USB, and the DIN output of that cable is attached to a synth module or something similar?  What is generating the MIDI notes (a MOD plugin, external MIDI controller, etc.)?  The notes Are received for some time, but fail later?  Does a reboot, fix the problem (for a similar duration)?  It would be helpful to see a screenshot of the MIDI config as seen from the MOD-UI (lower right corner).

protectorateindustrial, 2023-10-24:
I'm connecting the usb midi interfaces directly to the rpi is usb ports. Anyway I was able to fix this by uninstalling anything jack related (including the libraries) and reinstalling jack. I was also able to reproduce the problem again by installing the dependencies the are listed in the pi-stomp mod install script. I guess theres a jack dev library in those deps that cause the usb midi out to drop out after like 50 events.