TreeFallSound/pi-stomp #96: Strange JACK issues

rusk911, 2024-01-06:
Jack is running, mod-ui is working audio is passing through pedalboard, but test shows error

 16:35:52|root        |ERROR |Make sure jackd isn't running  

I\d ignore this but I have just tried zita-j2n and it fails with errors:

`Cannot connect to server socket err = No such file or directory
Cannot connect to server request channel
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
Can't connect to Jack, is the server running ?
`
strange. same things happens with jack's native jack_lsp -c
I just wanted an option to send audio directly to Ardour running on desktop PC over LAN

rusk911, 2024-02-04:
Continue investigating things. Paul Davis wrote "It’s “difficult” in part because we never intended JACK to be a system service."
May be this is the reason?

rreichenbach, 2024-02-04:
Possibly.  I'm near certain MOD was not designed to share the Jack service.  Which base operating system are you running (bullseye, bookworm, etc.)?

rusk911, 2024-02-04:
bookworm

rusk911, 2024-02-04:
> MOD was not designed to share the Jack service

I beleave it could be one of most wanted features. Send both processed and DI streams to DAW for recording and later reamping and hear your pedalboard at same time. No need digital ins and outs, no need to send audio through DAC-ADC. Just play your rig and record it over network on your studio PC. Since zita-njbridge is available, may be would be better to add this to mod-ui with a text field for DAW IP address and port setting

rreichenbach, 2024-02-04:
I agree, would be very cool.

The pi-Stomp project doesn't officially support bookworm yet.  We're working on it, but lots has changed.  Can't promise that your specific issue would be any different, but if you did want to try something else to get it working, I'd recommend starting with bullseye, lite, 64-bit.

rusk911, 2024-02-04:
Btw, the pistomp installation guide worked well for me with audioinjector.
Then i tested it with maudio fast track and now with pisound hat.
Ewerything running fine on rpi4

rreichenbach, 2024-05-17:
Closing issue as the problem is with the bookworm distro which we don't yet support.  Until we do, use bullseye 64-bit lite as a base OS as the installation guide suggests.

closed 2024-05-17 by rreichenbach