TreeFallSound/pi-stomp #101: IQaudio Codec Zero only build

Klinenator, 2024-10-31:
Is it possible to get this working with the IQaudio Codec Zero only? And use the web UI for everything else?

rreichenbach, 2024-10-31:
Maybe expand more on what you mean by get THIS working.  If you mean you just want a MOD platform for audio, without any of the control features that pi-Stomp hardware and software allow, sure.  There may be a simpler solution out there, but pi-stomp does round up the pieces you need, so it might get you up and running quicker.

Pretty easy to just burn this beta build to an SD card and go:
https://github.com/treefallsound/pi-stomp/releases/

Since you don't have a display, you might want to disable/stop the main pi-stomp service so it's not eating up cpu cycles:
sudo systemctl disable --now mod-ala-pi-stomp

Klinenator, 2024-11-01:
I keep getting the error ALSA lib confmisc.c:165:(snd_config_get_card) Cannot get card index for MODDUO

Even though im using the IQaudio Codec Zero

rreichenbach, 2024-11-01:
Make sure you get this output from this command:
`aplay -l`
**** List of PLAYBACK Hardware Devices ****
card 0: IQaudIOCODEC [IQaudIOCODEC], device 0: IQaudIO CODEC HiFi v1.2 da7213-hifi-0 [IQaudIO CODEC HiFi v1.2 da7213-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0

It should be card 0.  If that all looks good, it would be worth checking for system errors by running:
`dmesg`

Klinenator, 2024-11-01:
I dont see any errors in dmesg

Klinenator, 2024-11-01:
> My output for aplay -l is
> card 1: vc4hdmi0 [vc4-hdmi-0], device 0: MAI PCM i2s-hifi-0 [MAI PCM
> i2s-hifi-0]
>   Subdevices: 1/1
> card 2: vc4hdmi1 [vc4-hdmi-1], device 0: MAI PCM i2s-hifi-0 [MAI PCM
> On Fri, Nov 1, 2024 at 12:04 AM Randall Reichenbach <

Also i discovered The Makefiles for mod-host and mod-ui are not working

-Bsymbolic needs to be added to both for LDFLAGS

rreichenbach, 2024-11-01:
If you're building your own, then you're on you own.  I gave you a link earlier for a prebuilt image:
https://github.com/treefallsound/pi-stomp/releases/

Use that and I'm near certain, you'll have a working system within 20 minutes.

Klinenator, 2024-11-01:
Oh perfect, I didn’t see that. Thank you

Klinenator, 2024-11-02:
You were 100% correct. Loaded the image and was up and running immediately

Thank you

rreichenbach, 2024-11-02:
Cheers!  Closing issue

closed 2024-11-02 by rreichenbach