TreeFallSound/pi-stomp #103: jack utils

rusk911, 2025-02-09:
Hello. I am trying to connect pistomp with my PC for recording via LAN using zita-njbridge. Seems any jack application cannot connect to jack running as systemd. Do you have some idea how to do that?

rusk911, 2025-03-14:
First of all, I have managed to run zita-j2n as a service and solved user space jack utils connection to jack server. There's an environment variable JACK_PROMISCUOUS_SERVER=jack in jack service. Adding this to .bashrc allowed connections to jack socket from user space.

BTW zita-j2n still not working. Seems mod-host see new hardware and adding 2 more output ports but with wrong names. Trying to connect it through mod-ui in browser resulting these log lines:

```
Mar 13 14:31:26 pistomp jackdrc[367]: Unknown destination port in attempted (dis)connection src_name [effect_6:out_l] dst_name [system:in_1]
Mar 13 14:31:26 pistomp jackdrc[367]: Unknown source port in attempted (dis)connection src_name [system:in_1] dst_name [effect_6:out_l]
```

But jack client have different name:

```
pistomp@pistomp:~ $ jack_lsp -c
....
zita-j2n:in_1
zita-j2n:in_2
....
```

rreichenbach, 2025-03-16:
I've not worked with zita either.   Since these issues have nothing to do with this software repo, I'm closing them.  You might find better help from either the MOD forum:
https://forum.mod.audio/
OR pi-stomp forum:
https://treefallsound.com/forum

closed 2025-03-16 by rreichenbach