TreeFallSound/pi-stomp #104: USB audio gadget

rusk911, 2025-03-15:
Still trying to deliver some digital stream to PC without additional da/ad conversion.
Enabling usb gadget like is described in [mod-audio wiki](https://wiki.mod.audio/wiki/Audio_Through_USB) makes mod-host 100% cpu load and jack errors flood in syslog.
pi-stomp core on raspberry pi 4.
Please any advice?

rreichenbach, 2025-03-16:
I've not tried this.  Realize that those instructions are for MOD hardware and it sounds questionable at best.  38ms of latency?  I give up when it's much more than 15ms.

Anyway.  If you want to see what's going on with mod-host, run:
journalctl -fu mod-host

rusk911, 2025-03-16:
Thank you for response.

Nothing interesting in journalctl about mod services. Jack service cannot connect to anything

```
jack.service - JACK2 Audio Server
     Loaded: loaded (/lib/systemd/system/jack.service; enabled; vendor preset: enabled)
     Active: active (running) since Sat 2025-03-15 20:17:04 CET; 12h ago
   Main PID: 363 (jackd)
      Tasks: 5 (limit: 1010)
        CPU: 1min 50.875s
     CGroup: /system.slice/jack.service
             └─363 /usr/local/bin/jackd -t 2000 -R -P 95 -d alsa -d hw:0 -r 48000 -p 256 -n 2 -X seq -s

Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_7 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = mod-monitor was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_6 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_10 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_11 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_9 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_7 was not finished, state = Running
Mar 16 08:40:13 pistomp jackdrc[363]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 16 08:40:13 pistomp jackdrc[363]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = mod-monitor was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_6 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_10 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_11 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_9 was not finished, state = Running
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_7 was not finished, state = Running
Mar 16 08:40:13 pistomp jackdrc[363]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_10 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_9 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_7 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackAudioDriver::ProcessGraphAsyncMaster: Process error
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = mod-monitor was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_6 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_10 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_11 was not finished, state = Triggered
Mar 16 08:40:13 pistomp jackdrc[363]: JackEngine::XRun: client = effect_9 was not finished, state = Running
```

MOD developers [pointed](https://forum.mod.audio/t/zita-njbridge/12389/4?u=dmitry_sushkov) me here.

Haven't read about 38ms latency anywhere... I beleave it's related to windows drivers, not to jack. Since I use Linux and jack on my desktop and on rehearsal laptop it shouldn't be an issue. MOD developers recommended to use USB audio gadget instead of jack-netmanager. But in pi-stomp I can't get any of this working.

rusk911, 2025-03-16:
> 38ms of latency? I give up when it's much more than 15ms.

Its full roundtrip latency under shitty windows. RPi4 with sound interface (i use pisound hat, which uses i2c) does 16.942 ms total roundtrip latency with default 256 samples (just measured with jack_iodelay). Desktop [sound interface also have around 14 ms with 128 samples buffer size](https://linuxgamecast.com/2020/10/interfacing-linux-focusrite-saffire-pro-40/). So 30 ms total. Add wireless guitar plug and IEM system and you get far out of usable latency. USB class compliant device just passing digital audio streams without resampling and AD/DA conversion shouldn't have delay more then 0.5 ms, thats why I'm trying this. I want to go wireless and trying to reduce any piece of latency.